OllisGit/OctoPrint-FilamentManager #116: Plugin doesn't install on Ubuntu 22 VM

daimon82, 2023-05-08:
**Describe the bug**
The UI install ends with "Done! and Install Unknown /checkmark/" after octoprint restarts plugin is not installed and operation can be repeated.
Install using pip using the PI user ends with:

 Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 255
  ╰─> [1 lines of output]
      Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under?
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

**Steps to reproduce**
1. Install ubuntu 22.04.02 on a Virtual Machine (proxmox in my case)
2. Click on plugin manager
3. Install Filament Manager
4. See error

**Expected behavior**
Inatall Complete

**Did the same happen when all other 3rd party plugins are disabled?**
Every other plugin installs successfully just like on RPI.

**Log file**
[octoprint.log](https://github.com/OllisGit/OctoPrint-FilamentManager/files/11419859/octoprint.log)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

jattie-ire, 2023-05-16:
I can confirm this, no plugin after install, like it just never installed, looks like it installs, but nothing appears after.

jattie-ire, 2023-05-17:
Downgrading to Python3.9 restores the Filament manager and Spool manager functionalities to a working state on a Linux instance.

jattie-ire, 2023-05-24:
Hi Olli,

just to keep you in the loop, it looks like a SQLAlchemy version issue with the Linux distributions and if not explicitly installed in the octoprint venv. So by updating it seems to resolve the issue.

https://github.com/sqlalchemy/sqlalchemy/discussions/9831