OllisGit/OctoPrint-FilamentManager #119: Can't install FilamentManager after Linux distro update

Chainz11, 2023-06-19:
**Describe the bug**
After updating the Kubuntu distro which was described in [this thread ](https://community.octoprint.org/t/cant-install-plugins-after-linux-kubuntu-distro-update/51730) The only difference is that now I'm running octoprint 1.9.0. 
The plugin Filamentmanager seems to install but when it finishes the instlalation it shows sucess plugin unknown notification and after restart the plugin doesn't show in available plugins. When I try to install it again it says ....uninstalling filamentmanager .... installing.... and if I continue the whole process repeats 

**Expected behavior**
it should show up as installed plugin and I should be able co control its functions through the interface

**Did the same happen when all other 3rd party plugins are disabled?**
If not which plugin is causing the issue?

**Log file**
`023-06-19 10:23:25,958 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/OllisGit/OctoPrint-FilamentManager/releases/latest/download/master.zip
2023-06-19 10:23:26,359 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager
Traceback (most recent call last):
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint/plugin/core.py", line 1290, in _import_plugin
    module = _load_module(module_name, spec)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
           ^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 721, in _load
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint_filamentmanager/__init__.py", line 17, in <module>
    from .data import FilamentManager
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint_filamentmanager/data/__init__.py", line 13, in <module>
    from sqlalchemy.engine.url import URL
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/__init__.py", line 9, in <module>
    from .sql import (
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/__init__.py", line 8, in <module>
    from .expression import (
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/expression.py", line 34, in <module>
    from .functions import func, modifier, FunctionElement, Function
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/functions.py", line 11, in <module>
    from . import sqltypes, schema
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/sqltypes.py", line 17, in <module>
    from . import elements
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/elements.py", line 16, in <module>
    from . import type_api
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/type_api.py", line 16, in <module>
    from .base import SchemaEventTarget
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/base.py", line 49, in <module>
    class _DialectArgView(collections.MutableMapping):
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'collections' has no attribute 'MutableMapping'
2023-06-19 10:23:26,384 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.6.13) did not pass check, not loading.
2023-06-19 10:23:34,624 - octoprint.plugin.core - ERROR - Error loading plugin filamentmanager
Traceback (most recent call last):
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint/plugin/core.py", line 1290, in _import_plugin
    module = _load_module(module_name, spec)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
           ^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 721, in _load
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint_filamentmanager/__init__.py", line 17, in <module>
    from .data import FilamentManager
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/octoprint_filamentmanager/data/__init__.py", line 13, in <module>
    from sqlalchemy.engine.url import URL
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/__init__.py", line 9, in <module>
    from .sql import (
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/__init__.py", line 8, in <module>
    from .expression import (
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/expression.py", line 34, in <module>
    from .functions import func, modifier, FunctionElement, Function
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/functions.py", line 11, in <module>
    from . import sqltypes, schema
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/sqltypes.py", line 17, in <module>
    from . import elements
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/elements.py", line 16, in <module>
    from . import type_api
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/type_api.py", line 16, in <module>
    from .base import SchemaEventTarget
  File "/home/kamen/OctoPrint/venv/lib/python3.11/site-packages/sqlalchemy/sql/base.py", line 49, in <module>
    class _DialectArgView(collections.MutableMapping):
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'collections' has no attribute 'MutableMapping'
2023-06-19 10:23:34,646 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.6.13) did not pass check, not loading.
2023-06-19 10:23:34,717 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint.`
**Screenshots**
[image]

wizardkitty, 2023-07-05:
this is happening on ubuntu also. can confirm.

wizardkitty, 2023-07-07:
I recommend using this https://github.com/bitsy-ai/OctoPrint-FilamentManager/tree/fix-python-10-compat

if you wish to install it use this as a url https://github.com/bitsy-ai/OctoPrint-FilamentManager/zipball/fix-python-10-compat

I can confirm it works.